makerbase-mks/MKS-DLC32 #200: MKS-DLC32 WIFI Access Point

BigJuicee, 2022-05-06:
What is the configuration for the Wi-Fi access point?
This is what I have right now, but I don't know what's wrong or what else I need.

//--WIFI configuration---//
$ESP115=ON
$ESP110=AP
$ESP131=8080	
//wifi name
$ESP105=MAKERBASE3D 	
//wifi password，：[ESP106]makerbase3d
$ESP106=makerbase3d	
$ESP107=192.168.0.1
$ESP108=1
$ESP120=ON
$ESP121=80

abinoer, 2022-05-06:
[image]
You just need to change the wifi name and wifi Password in this file. However, you can use MKSLaserTool to connect to WiFi, which is much easier and faster

BigJuicee, 2022-05-06:
How would I use the MKSLaser Tool to connect to WiFi? I want to have it set up so it gives me an IP address but I can't figure it out.

abinoer, 2022-05-06:
[image]
This software, after you download and install, select the correct COM port to connect, enter the name and password to connect to WiFi, there is a button to obtain IP, very simple, you try

BigJuicee, 2022-05-06:
Could you give me an example of what the code would look like?

abinoer, 2022-05-07:
[image]
[image]

BigJuicee, 2022-05-07:
Is that in the configuration? Could you send the txt file? I need something to work off of that would get me in the right direction at least. The Wi-Fi settings in the dlc_cfg.txt wouldn't work for me.

abinoer, 2022-05-07:
[image]
[image]
1. Copy the modified configuration file to the TF card, insert it into the motherboard, wait for about two minutes, and then connect to WiFi after restart
2. Connect using MKSLaserTool
3. Connect by command
All of the above methods can be connected to WiFi, You can connect in another way,the most convenient is through MKSLaserTool

closed 2022-05-07 by abinoer

3DSmitty, 2022-05-11:
@abinoer this makes no sense. Anyone familiar with programming ESP32 knows "STA" mode  means it is a client station on a wifi network. I believe the original question is how to set this up as a wifi access point as FluidNC does by default. Does the DLC32 support "[ESP110] AP" setting and if so what would the whole configuration look like?