makerbase-mks/MKS-DLC32 #201: MKS False promises

cfox1028, 2022-05-06:
It's been nearly a year from when they promised to release the source code.

They can't even add a simple Z button to the laser firmware.

Anyone reading this - don't get your hopes up for any meaningful support on these boards.

daxliniere, 2022-05-08:
Switch to FluidNC (but lose LCD function)

MTRatcliffe, 2022-05-08:
Would be nice to see this shared and to see what comes from it [ ie leveraging the blooth of the esp32 to accept a ps3 remote as a pendant].

cfox1028, 2022-05-11:
> Switch to FluidNC (but lose LCD function)

That defeats the purpose of having the LCD.
Also Fluid is pretty far behind the likes of GRBLHAL so I wouldn't go that route anyway.

The source should be open so we can have our own configurations. I would like to keep the LCD functionality but adjust things myself.

hallskelet, 2022-05-26:
Buttons for moving Z+, Z- and reset Z=0 are also needed for the laser. this is how the table moves to adjust the focus.

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2022-11-15 by makerbase-mks