makerbase-mks/MKS-DLC32 #204: MKS DLC32 with TS35 display - CNC

dmordred, 2022-05-09:
I purchased a CNC powered by a MKS DLC32 from aliexpress under the promises it'd carve a relief in wood, but I'm yet to be able to make it run for more than 10 minutes. It simply stops. Tried with direct connection and from the sdcard. Apparently it freezes when the code changes layer, and considering the amount of people complaining about the firmware, something is fishy on that side. 
I must ask: 
-Does it support multiple layer engraving? As in carving a 3d piece out of wood? or just a 2d cut?
-Is there any firmware that can be used to achieve such 3d carving capabilities? GBRL?

abinoer, 2022-05-10:
Can you send over your carving files and we'll find out why

dmordred, 2022-05-10:
thanks for the prompt reply, but I've already found the problem. The default settings on the dlc_cfg file set the maximum rate way too high for my CNC to handle, making the x/y de-synch and hitting the endpoint. Updated the firmware to the latest cnc and set it to 500 and now it's working!

closed 2022-05-16 by abinoer