makerbase-mks/MKS-DLC32 #205: incorrect/unrequested homing action when using carve button.

miamijerry, 2022-05-10:
Using the CNC firmware, when playing a gcode file by the 3.5" display. At the end of the file the MKS DLC32 tries to home, but does not work well, z axis does not return completely to zero and Y tries to go past zero (Strips GT2 belt). Selecting either home buttons on the 3.5 display works perfect. Playing the same gcode file by ugsplatform, the machine stays where the last code put it, which is really what should happen. The play/Carve button on the 3.5" display is creating an incorrect homing request at the end of a gcode file?

abinoer, 2022-05-13:
I don't understand what you mean. Do you mean that there is no need to automatically return to the position after carving?

miamijerry, 2022-05-13:
After carving, there is some a hidden function (Not in the gcode) which is trying to home the machine, but it is using incorrect homing information. Y is trying to go past zero, which was the starting point. Z moves short of reaching zero (The starting point). XY and Z home button on the touch screen work perfectly.

Would prefer the ability to turn this function OFF, as the fact that it tries to return Y past zero (Not possible on my machine) make this a big problem.

miamijerry, 2022-05-13:
> Do you mean that there is no need to automatically return to the position after carving?

Not if it uses incorrect home information

abinoer, 2022-06-25:
Fixed, please update firmware

closed 2022-06-25 by abinoer