makerbase-mks/MKS-DLC32 #206: new CNC firmware problem

kweilim98, 2022-05-12:
hi MKS, I have some problems would like to say about the lasted CNC firmware.
1. why is the buzzer always on when touching the screen panel?
2. why is the WebUI feature still the same and not added in the probe feature?

https://user-images.githubusercontent.com/66898187/167990882-5eacdf86-89f7-4c96-ae61-acd406afc1b1.mp4

[image]

kweilim98, 2022-05-12:
the buzzer should be on when the axis touches the end-stop or the stepper motor was enabled ($4=1) rite? we no need the function as the child's play toy when the press comes out with the sound

abinoer, 2022-05-13:
We have received your feedback and will fix it

kweilim98, 2022-06-25:
@abinoer  when this bug will be fix?

abinoer, 2022-06-27:
The problem with the buzzer has been fixed. There are still problems with the probe function on the WEB, which is still being fixed

kweilim98, 2022-06-27:
> The problem with the buzzer has been fixed. There are still problems with the probe function on the WEB, which is still being fixed

Alright

kweilim98, 2022-06-27:
> The problem with the buzzer has been fixed. There are still problems with the probe function on the WEB, which is still being fixed

你懂中文吧？你是这个dlc32的负责人吧？可以加个微信吗？因为我也不好意思一直去打扰羽中君，如果不介意的话，留下你的微信号，我去加你，方便交流

abinoer, 2022-06-28:
您可以加入我们的QQ群（650031716）进行交流

kweilim98, 2022-06-28:
我这里是海外的，注册不了qq号，所以才逼不得已添加你的微信号的，希望你可以理解哥

kweilim98, 2022-07-09:
> The problem with the buzzer has been fixed. There are still problems with the probe function on the WEB, which is still being fixed
嗨你好，根据你说这个蜂鸣器的问题已经解决了。可是当我上传最新的cnc firmware时，蜂鸣器还是一样没功能。只是添加了蜂鸣器的开关而已。