makerbase-mks/MKS-DLC32 #208: Power input

Hargard339, 2022-05-15:
Hi. I have a issue of insufficient power(Amps) getting into my board(MKS DLC V2) as the input is a 5.5 X 2.1/2.5  input Jack rated at 1 Amp.
Is there anywhere I can put a power input into the board other than the power input Jack ??
I am using MKS2209 drivers rated at 2 Amps. per drives. 
The jack plug is catching fire or just melting. I need a minimum of 4 Amps @ 24 Volts.
Or is it a case of binning the board and getting a new one capable of handling the power input.
Regards Hargard.

abinoer, 2022-05-16:
1.  Do you catch fire when you connect the power supply without connecting anything? Is there a short circuit somewhere?
2. 2209 seems to support only 1.8A maximum continuous voltage, the maximum peak current is 2.5a, you set it to 2A, there may be A problem, you reduce the current is there A fire situation?
3.  if the motherboard is not short circuited, you may need to connect the external drive, because your requirements of the drive current is too large

Hargard339, 2022-05-16:
No Its Only when the machine is running. and its the jack plug that heats up. The board has been tested and shows no shorts. 
If 3 Steppers are running then it is issue.   
Drivers are set at 1 Amp only. 
I have managed to make my own jack plug that will carry 5amps by 3D printing and brass connectors. Its a Fix not a solution to the issue.

abinoer, 2022-05-16:
[image]
We have replaced it with this kind of interface. Is your DLC V2 this kind of power interface?