makerbase-mks/MKS-DLC32 #212: $Homing/Cycle1=Z

william1370, 2022-05-19:
CNC w/ Hard limit switches
Took me forever to figure out why my Z axis wasn't homing... finally found this setting "$Homing/Cycle1=0"
When set to "Z" my Z axis does home but it causes an alarm immediately upon hitting the limit switch.
I've changed many settings but haven't seen an improvement.
Also... why does Z home last??? Seems like a major clearance issue. Is this just a grbl thing? Because FluidNC accommodates for it.

daxliniere, 2022-05-26:
That's odd to read, since I have used the $homing/cycle settings and they worked well.

At terminal (UGS, etc), send these:
$Homing/Cycle0=Z
$Homing/Cycle1=XY

daxliniere, 2022-05-27:
> CNC w/ Hard limit switches Took me forever to figure out why my Z axis wasn't homing... finally found this setting "$Homing/Cycle1=0" When set to "Z" my Z axis does home but it causes an alarm [...]

Okay, I have found the "issue". In FluidNC, Cycle 1 is the first cycle, Cycle 2 is the second, etc. In GRBL_ESP32 (which is what DLC32 runs), Cycle0 is first , Cycle1 is second, etc.