makerbase-mks/MKS-DLC32 #213: LOGO CUSTOM

kweilim98, 2022-05-20:
when the logo can be custom and open source? Just like the tft can use the mks tool to custom the icon. When dlc will be?

aviador58, 2022-07-14:
Has it been possible to solve this issue? I want to personalize my plate.

kweilim98, 2022-07-15:
> Has it been possible to solve this issue? I want to personalize my plate.

Not sure, I m still waiting too

Medina-ui, 2022-07-15:
Hi, No updates I can see at the moment! Still waiting too. Almost 2 Months since it has been move to "In progress" and I have several boards, tft's and machine enclosures waiting for this. Thinking about returning to old one.....

aviador58, 2022-07-15:
gracias por contestar.

aviador58, 2022-07-15:
I don't understand how they don't open the code more to be able to get more out of these boards, we are going to have to go back to the usual open source boards, because they are updated much more to the new needs of the users.

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2022-11-15 by makerbase-mks

sharknetwork24, 2023-07-12:
The solution to change the image to your display.

https://github.com/makerbase-mks/MKS-DLC32/issues/390

burak1000, 2023-08-01:
i tried many times this method but i cant do it. it's simply not works

Medina-ui, 2023-08-01:
I burak1000. Send me the image and tell me what version of firmware you need. I can do it for you