makerbase-mks/MKS-DLC32 #214: It is inconvenient to position the laser on the blank

hallskelet, 2022-05-21:
The problem is that my machine has a large working field (1m by 0.5m), and the positioning buttons on the screen give steps of 0.1, 1 and 10mm. This makes them very inconvenient to use.
Could you change the positioning steps to "1, 10, 100mm"?
Or add more buttons (eg 50 and 100)?

solawc, 2022-05-21:
It may be difficult to decide. The size of the machine used by everyone is different. Some people use small machines, such as (120 * 120), and some use such large machines like you; Maybe there are other good setting suggestions? Instead of making setting changes on a real personal machine

hallskelet, 2022-05-26:
Can this be done through the settings field in the web interface?
Then everyone will be able to make such a setting for their car.
Or will it be hard to do?

Lasermax3000, 2022-06-07:
Why can´t you change the HMI, that whan you push for example X+ Button the Laser drives till you release the Button?
Some fine tuning Buttons for small stepps and a continuos jogging Mode would be great, may be you can add a Parameter in the Config or HMI for Adjusting the Speed of continuos jogging.

abinoer, 2023-01-09:
Sorry, we don't have the energy to modify the custom requirements for each DIY customer. The firmware has been open source, please modify it yourself

closed 2023-01-09 by abinoer