makerbase-mks/MKS-DLC32 #215: Critical error when working with the machine over the network

hallskelet, 2022-05-21:
If I start the machine from a browser or via usb. That all works well.
Only now the board does not understand that it is controlled through an external connection.
If I press something on the screen, it either reboots or resets the program.

Could you make a screen lock if external control is disabled or make a screen change with external control leaving only the "stop" button there to stop.
And return the screen to normal when external control is disabled.
This very unexpected and annoying problem resulted in several parts being damaged.

geogarden, 2022-05-22:
I have a similar error. When the laser engraves via USB and I open the engraver's website, the machine stops and "freezes". I have to do a HARD RESET. Of course, work to be fired. A bit annoying bug. Could you turn off the WIFI with a "button"?

abinoer, 2022-05-24:
Does that mean manipulating the screen while using the WEB? Currently, multiple controls are not supported at the same time,

geogarden, 2022-05-24:
> Does that mean manipulating the screen while using the WEB? Currently, multiple controls are not supported at the same time,

Not. Just open the web page "by accident" while working, and the engraver freezes immediately. Only a hard reset of the device helps.

hallskelet, 2022-05-26:
> Does that mean manipulating the screen while using the WEB? Currently, multiple controls are not supported at the same time,

Could you make it so that if I started working via USB, then it would not be possible to enter the WEB interface or type something on the screen.
To prevent accidental simultaneous control.
As for example done in EDGE TX

geogarden, 2022-05-26:
That would be a good feature. If USB is not WIFI and vice versa.

abinoer, 2022-06-02:
> Could you make it so that if I started working via USB, then it would not be possible to enter the WEB interface or type something on the screen. To prevent accidental simultaneous control. As for [...]

Sorry, that's not possible

closed 2022-06-02 by abinoer