makerbase-mks/MKS-DLC32 #216: TTL port with pwm signal convert to analog 0-10v

kweilim98, 2022-05-23:
Hi , I am CNC user, I am facing the problems when using the pwm signal from TTL port with CNC firmware, when the power on, the converter module always on . No matter I try using m03 or m05 to variable it.  And the input pwm always in 0v. Anyone can help me?
[image]
[image]
[image]

abinoer, 2022-05-24:
1. Which signal of 0V do you measure specifically? What exactly did you type in? Does TTL output when you input an instruction without receiving anything? What position is 0V that you measure
2. What is the voltage requirement of the input signal of your converter module?
3. Why are TTL interface and spindle interface connected at the same time?
[image]

kweilim98, 2022-05-24:
[1.pdf](https://github.com/makerbase-mks/MKS-DLC32/files/8760365/1.pdf)

> 1. Which signal of 0V do you measure specifically? What exactly did you type in? Does TTL output when you input an instruction without receiving anything? What position is 0V that you measure
> 2. [...]

kweilim98, 2022-05-25:
https://user-images.githubusercontent.com/66898187/170189947-4bb050f1-7d9d-4bfd-b993-f632a432d2d5.mp4

kweilim98, 2022-05-25:
Have anyone know how to variable the spindle speed with pwm signal? As video show the when M03 S650, the spindle already in 10k and above

abinoer, 2022-06-01:
[image]

Connect as required by your module, paying attention to the jumper of the short cap. You can consult the business selling modules

kweilim98, 2022-06-19:
> > [image]
> > Connect as required by your module, paying attention to the jumper of the short cap. You can consult the business selling modules
> 
> me too.. the set spindle speed 30% to 100% the out put pwm is the same, full power voltage 5v in ttl pin. set 5% pwm power start goes down to 50% 2,8v. been tried to set $28 to different numbers its change pwm khz and bits..and the out come is more weird... anyone facing my problem

Hi bro, do u get it to solve the problems the spindle was not simultaneously in the rpm? If u have solve it can u share here?

dsp8bit, 2023-10-27:
Hi, I have the same issue. I checked with the oscilloscope and the DLC32 board only accepts a maximum spindle speed of S30=1000. Hence the PWM's frequency $28=400Hz allows to change the duty cycle (S250 = 25% , S500=50% or S900=90%).  I change the maximum spindle speed beyond 1000, the board doesn't care.