makerbase-mks/MKS-DLC32 #222: MKS-DLC 32 firmware V2.0.8_H35_20220105_N.bin problem

msmahm48, 2022-05-27:
Hi
I have updated my MKS-DLC 32 V2.1 firmware to the V2.0.8_H35_20220105_N.bin . The homing works OK for all the three axis. But the laser stored working. I checked the $32 and its 1. Any suggestion on how to solve this problem?
Thank you.

abinoer, 2022-06-01:
Laser not working? Is the wiring wrong? What is the motherboard model you are using, is the firmware in board_v1.0 or board_v2.0?

msmahm48, 2022-06-01:
The motherboard is  MKS-DLC 32 V2.1 and I am using the firmware in  board_v2.0. The laser wiring is correct because I have checked it with other firmware on the same board.

abinoer, 2022-06-02:
[image]
How do you turn on the laser? Will the indicator light turn on after sending the command "M3 S1000 G1 F1000"?

msmahm48, 2022-06-02:
I am using Lightburn. The indicator is off always even after sending the command "M3 S1000 G1 F1000".
This is my setting:
$25=1000.000
$26=250.000
$27=1.000
$28=1000.000
$30=1000.000
$31=0.000
$32=1
$38=1
$40=1
$100=80.000
$101=80.000
$102=520.000

abinoer, 2022-06-02:
You re burn the firmware, if the indicator light still does not light after sending instructions, there should be a problem with the motherboard, then you contact customer service

closed 2022-06-02 by abinoer

msmahm48, 2022-06-02:
Hello
You are right. I don't know how that has happened. The firmware file V2.0.8_H35_20220105_N.bin that I have was something wrong with it. I downloaded a new copy and it is working as expected. 
Thank you so much.