makerbase-mks/MKS-DLC32 #223: Z axis limit switch not working? (3.5" lcd, laser version)

msmahm48, 2022-05-31:
Hello, 
I'm trying to use a limit switch for the Z-axis but the board doesn't seem to react to it, no alarm is triggered.
I have v2.0 of the board with the 3.5" LCD and the most recent laser firmware (V2.0.8_H35_20211222_N.bin).
I have tried the firmware for the 2.4" (V2.0.8_H24_20220107_N.bin) it works perfectly, but the LCD stopped working.
If you can update the firmware (V2.0.8_H35_20211222_N.bin) I would really appreciate that. 
Thank you

abinoer, 2022-06-01:
Isn't it a newer firmware version?
The command to turn on this feature: $Homing/Cycle1=Z
The command to turn off this feature: $Homing/Cycle1=0

abinoer, 2022-06-02:
How do you determine unavailability? Isn't there a copy of V2.0.8_H35_20220105_N.bin? It's been tested and it works

msmahm48, 2022-06-02:
Hello
You are right. I don't know how that has happened. The firmware file V2.0.8_H35_20220105_N.bin that I have was something wrong with it. I downloaded a new copy and it is working as expected. 
Thank you so much.

closed 2022-06-06 by abinoer