makerbase-mks/MKS-DLC32 #225: Controlling MKS dlc32 by WIFi

PA3CCE, 2022-06-02:
I succeeded to open the WIFi port on my Windows 11 PC by using the  MKSlLasertool software.
What software can I use now to control my MKS DLC32-TS35 normal CNC device via WiFi ??
The device is working well stand-alone and wich LaserGRBL etc . via USB,  but I prefer to use a wireless connection to the device by my PC.

PA3CCE, 2022-06-07:
Through the web yess, but in my environment I see no connectable item on the LCD ,they all are secured.
I want to connect my PC to my DLC, but there is no CNC softare I know of that can connect over WiFi, all of them connect over USB.
I have odered a USB to WiFi adapter, I hope this can solve the problem ?

PA3CCE, 2022-06-16:
No that was not solving the problem, I just got a  ssecond identical LAN (LAN2), so no USB connection to WiFi.
Sorry, Im not familar with local networking.
Some time ago I assked why Bluetooth is not implemented, that would solve my problem, I don`t know how to use WiFi .

Witawat, 2022-10-20:
can connect wifi Tibbo
[image]
[image]
[image]