makerbase-mks/MKS-DLC32 #226: Flame detection function

msmahm48, 2022-06-03:
Hello
I am using the MKS-DLC32 V2.1 as a laser engraver. I want to attach a flame detector connected to the probe input. How can activate this function? Any diagram and setting?
Thank you

Lasermax3000, 2022-06-07:
I think you have to update the dlc32_config.txt, because that´s not a standard setting in the normal grbl Firmware. 
That might be part of the "custom" Firmware Part of MKS. But you can try to change the setting by sending $37=1 in the cli. 
If this doesn´t work -> update dlc32_config.txt

msmahm48, 2022-06-07:
I have tried both. But the problem I think is that the $37 was not set in the firmware. Because when I send S37=1 the board return this error::
$37=1
error:3
Grbl '$' system command was not recognized or supported.

buzzshot, 2022-09-19:
Not Sure if read the Doc for the wiring wright but it says 5V to S and that didn't work. Used G to S along with reflashing the dlc_cfg.txt to a micro SD plug it an after a min or two i restarted and everthing works.

Here's the part i edited also got the buzzer working by doing this so know if there is a fire
//---Protection function setting----//
//(Laser-mode enable)	
$32=1
//(Flame  1:Turn on,0:close) 
$37=1 
//(Beep 1:Turn on,0:close)  
$38=1

msmahm48, 2022-09-21:
Which firmware you are using?

Ams-Laser, 2023-03-06:
hello, i have firmware V2.1.0(8M.H24.20220511) i would like to tr other firmwares but i cannot find this original firmware download,  is there a download for it ? is there any newer firmware ?

Ams-Laser, 2023-03-09:
> 
> [image]
> 
> You have to set Parameter  $37 = 1 which activates the Flame detector.
> 
> 

Hello, do you know what the "gyro" función is for?

Thanks

Kuthjudas, 2023-09-03:
Hello, I am new here!
I have a laser engraving machine on which I mounted an MKS DLC32 board and which I have been working with. Now I would like to adapt a flame detector to it. From what I read in the manual, it is possible to activate the function using the "Probe" port.
Following some instructions I saw here I've already tried activating the port ($37=1) but the function remains inactive by simulating tripping using the flame detector, no error appears or interrupts the running job.
Below is an attached image of the display for you to analyze and try to understand the reason.
[image]
I apologize for my English, but it's very translator-based.

Kuthjudas, 2023-09-05:
Thank you very much for your prompt reply. I thought the answer would take a lot longer.

I must understand then, from your answer, that the "H24" in the file name refers to the 2.4 inch display and the "H35" to the 3.5 inch display, correct?
If so, I will effectively not be able to use the flame detection function.

>Read my post from 28 sept 2022 regarding the SD card with that firmware (four posts up!) [image]
This was the reason because I post an display image. ;)

dart1280, 2023-09-06:
That is how I understand it, yes. I would encourage you to try the H35 firmware. The only difference between the H24 and the H35 is the screen size so it should work except for the screen not displaying correctly. I am not sure how the screen will be wrong, perhaps it is just the icon sizes?  There is a guide and the software needed for flashing the DLC32 in the archive I listed - it is quite straightforward.

Kuthjudas, 2023-09-08:
I'm tempted to order just the 3.5 display and mount it on the board and then update the firmware. But I'm not sure if the 3.5 display is compatible with this board.

dart1280, 2023-09-08:
It's the same board.  Ensure you order the TS35 and not the TFT35..  Mine is a MKS TS35-R V2.0 and the sensor I used was one of these  [https://www.aliexpress.com/item/1005005644237065.html](url) using the digital output pin.

Kuthjudas, 2023-09-08:
If it is the same board that's cool. 
I' am searching by MKS TS35 display but I may try to flash the H35 firmware on this board... 
The board alone is sold without the display, so I believe it works even without the display.

This is the flame sensor that I have and intend to put it to work:
https://www.electrothinks.com/2021/01/KY-026-flame-sensor-module.html
Using the digital output of course.
Very similar to that one you share.

Kuthjudas, 2023-09-17:
Hello, I flashed the V2.10_H35_2022_0621_N_ZX_001 firmware and this does not solve the problem. 
After flashed the firmware I had to load the previous settings in order to work with the machine.
Before this I verified that, using MKS Tool, the $37 parameter did not appear but I also realized that few parameters were being shown.
[image]
Then using Lightburn tool (console) I ran the command to show all the parameters and $37 still didn't appear but others appeared that the MKS tool doesn't show.
[image]
Running the command to check the parameter value I get an unrecognized or unsupported command error.

After this I connected the display to the board just out of curiosity. You can see that the content is different but the lines on the display run over each other (as expected).
[image]

Right now the display is not the problem. I can work without it.
What have I done wrong?
Supposedly the card should have the port active, depending only on the value of $37, or am I wrong?
[image]

dart1280, 2023-09-19:
The sensor I used was a 3v3 device, you appear to have plugged yours into 5v
[image]

My flame sensor has the gnd and signal wires (purple and grey) going to the probe connector but the power feed going to the 3v3 from the I2C socket (white wire). The three red wires go to a relay board that uses the 5v feed.

What can I say .......  it works.

https://github.com/makerbase-mks/MKS-DLC32/assets/79755245/bec7c11d-ebea-4f04-9446-e33d4b1c795d

I use LaserGRBL and don't get the same warnings about unsupported commands, maybe it's a Lightburn thing... try LaserGRBL... it's free and does a good job without the complexity of Lightburn.

dart1280, 2023-09-19:
....Also, my dlc_cfg.txt file

[dlc_cfg.txt](https://github.com/makerbase-mks/MKS-DLC32/files/12657349/dlc_cfg.txt)

Kuthjudas, 2023-10-12:
Hello!!!
Thank you very much for all the information provided.
I've been absent from the laser hobby due to other priorities and for that I apologize but, in principle, I'm back. ;)
Effectively I have the sensor connected differently.
I will go back to using LaserGrbl and try to see the settings.
I'll give you news later. ;)

Kuthjudas, 2023-10-12:
Your configuration is clear in fact.
[image]

Kuthjudas, 2023-10-15:
So I have instaled the last version of LaserGrbl. Parameters related with Probe/Flame detector port still not visible...
[image]

I am wondering if the 3,5 display has any data that makes it possible...

dart1280, 2023-10-17:
Nothing to do with the display. The $37 flame mode setting is not a standard grbl parameter so LaserGRBL or Lightburn or any other standard grbl software wont list it. $37 is set in the dlc_laser.txt configuration file that is with the firmware.

Kuthjudas, 2023-10-18:
Hmmm ooookay, so what you're saying is that in the communication that is exchanged between the board and the display, it is not possible for the board to realize that it has a certain display connected to which the firmware does not match.

So in this case, I have two doubts. :)
- If the firmware is the same and the boards are the same model, why does the parameter $37 appear on one (yours) and not on the other (mine)? :)
- Will the parameter appear when installing newer firmware?
- (after all, there are three questions) Would it be possible to include or activate the $37 parameter by editing the txt file that comes with the firmware?

Thanks. ;)

dart1280, 2023-10-18:
The $37 parameter ONLY shows in your dlc_laser.txt file..... nowhere else, it is a piece of code written by dlc that the firmware interprets to function as a flame detector and sound the alarm when it detects IR. Presumably the firmware traps the instruction to prevent it being sent to the grbl software running on your computer to prevent 'unknown command' messages being generated.   AFAIK the $37 code written into the firmware is ONLY present in version V2.10_H35_2022_0621_N_ZX_001.bin firmware.

YES, you have to enable the flame detection by altering your dlc_laser.txt file to $37=1 and rebooting the controller with clc_laser.txt in the root directory of your SD card inserted in the controller.

Kuthjudas, 2023-10-19:
> YES, you have to enable the flame detection by altering your dlc_laser.txt file to $37=1 and rebooting the controller with clc_laser.txt in the root directory of your **SD card inserted in the controller.**

I always installed the firmware from computer and not from the SD card. Now I realize that you mention that a few times before and I don't pay attention to that detail. 
That will be my next task. 
Where can I find the "dlc_laser.txt" file
Maybe that's the reason because the parameter doesn't show. 

Thank you very much for your patience.

dart1280, 2023-10-19:
Oops.. it is actually called dlc_cfg.txt and it is located in the firmware download package from MKS-DLC32 github..

I renamed it on my system to temporarily differentiate it between laser and cnc applications but it needs to be renamed back to dlc_cfg.txt and in the root of the SD card for it to be accepted by the software.

Kuthjudas, 2023-10-21:
There is now light at the end of the tunnel!

After erasing what was on the board with "MKS laser tool" software, I removed the display (just in case), installed the firmware version V2.10_H35_2022_0621_N_ZX_001 and booted the board again.
I copied the dlc_cfg.txt file with $37=1 and $38=1 to the SD card and connected the board with the SD card inserted. I waited a minute or two to give the board time to configure itself, removed the SD card and turned it off. I installed the flame sensor and started the board again without the display on it.
When connecting on both the Lightburn or Lasergrbl, I already received an error from the flame sensor (even without a flame). Lighting a lighter in front of the sensor stops the error.
After turning off the board and assembling the display, I turned on the board and it emits a sound and you can see that the display is different from when there is no error. (I remember that the display is 2.4" and not 3.5 as it should be)
When I turn on the lighter in front of the sensor, the sound stops after a second or two. I still went to get the 3.3v I2C power supply like you have in yours but the result is the same.

In short, the flame sensor works in reverse. I assume that the flame detector you have mounted send the G signal
With the sensor removed, I confirmed that the alarm activates with the G signal.
To avoid having to change sensor, do you know of any way (by changing the dlc_cfg.txt file for example) to invert the activation of the alarm on the board from G to + (5 or 3.3v)?

dart1280, 2023-10-22:
No I don't. Your sensor is supposed to give a digital high signal when a flame is detected, have you tried twiddling the adjustment pot on the sensor board? Other than that you could try a new sensor, or. if the current sensor is reliable, make a simple NPN transistor inverter or if you want to get fancy - something like a SN74LVC1G04

Kuthjudas, 2023-10-23:
Hi!
Twiddling the adjustment only affects sensitivity, not this time (I didn't move it) but when I test it for the first time when I twiddle too much the sensor was always ON. 
For the NPN transistor or SN74LVC1G04 I have to bother my brother. He knows a lot of electronics.

This testing was under natural light. This sensor as two LEDs, one for power and other for active. When flame is detected that LED lights up. 
The other kind of light affect the sensor performance? I understand that de different type of light has different wave length, is that so critical for performance?
I will need to take this into account because when I go to work it will be set up in a small room with warm LED lighting.

Kuthjudas, 2023-11-11:
FINALLY!
Ok, the problem was the flame sensor signal output.

The problem was that the flame sensor I installed gave a positive output when it detected a flame and 0 in normal condition.
Unlike yours, which gave the output at 0 with flame and at positive in normal condition.
After analyzing the two sensors I realized that they both use the same LM393 chip.
So I bothered my poor brother (who knows about electronics) about "if it wasn't possible to change the output signal since the chip was the same". And so isn't it quite simple? It was enough to interrupt the original track of the output signal and solder a wire from the point that is 0 with flame to the output pin.
However, Santa Claus came early, and he already left me the 3.5" display :D

Summarizing this issue and pointing out what I consider important for my case:
- Purchase a flame sensor powered by 3.3 or 5v with output at 0
- The 3.5" display is not essential but it is useful.
- Delete the old firmware from the board and install firmware V2.10_H35_2022_0621_N_ZX_001.
- Copy the dlc_cfg.txt file with $37=1 and $38=1 to the SD card and connected the board with the SD card inserted.
After this, as long as the sensor is correctly connected to the "Probe" port, it should be working. In my case the first one didn't work because the sensor had an output with the opposite signal.

https://youtu.be/oL3GE_2x5NA?si=AoqTex7iJ-cTqH4a