makerbase-mks/MKS-DLC32 #229: Stepper Motors Stall If The Speed Is Over 1500

BigJuicee, 2022-06-08:
I have 2 NEMA 34 stepper motors and they stall if the speed is over 1500 mm/sec. I have the pulse and resolution to where they move the correct distance that I tell it to but I can’t get it to move as fast as a laser should. My stepper drivers are DQ860MA and they are set to 1600 pulse/rev and my X and Y axes travel resolution is 320. I don’t know what I have to adjust so the motors don’t stall and that they move the correct distance.

abinoer, 2022-06-09:
Will stop? What do you mean? Can't move faster than 1500? Could you possibly provide video to see what's going on

solawc, 2022-06-11:
Please ensure that:

1. Your power supply is stable

2. Do you have a 4988 driver in your hand? Have you tested the 4988?

BigJuicee, 2022-06-15:
[image]

https://user-images.githubusercontent.com/105027332/173854320-0706b7b6-7209-49c1-b766-69e57687cb6c.MOV

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

BigJuicee, 2022-06-15:
The 4988 driver is being used for the z-axis and the NEMA 34’s are being used with the drivers in the photos above.

BigJuicee, 2022-06-16:
When I change the pulse/rev it doesn’t move the correct distance in mm that I tell it to.

abinoer, 2022-06-16:
You can adjust the value of $100, $101, and $102 to get it right

BigJuicee, 2022-06-16:
Wouldn’t that be going up if the motor is set to 1600 for the pulse/rev?

abinoer, 2022-06-16:
If the distance is not correct, just adjust $100, $101, and $102

BigJuicee, 2022-06-16:
What should I have the current to the motors set to?

abinoer, 2022-06-16:
This needs to be set according to the load and motor parameters, so I can't give you any suggestions

closed 2022-06-16 by abinoer