makerbase-mks/MKS-DLC32 #231: Alarm 4 probe fail

kweilim98, 2022-06-14:
How to solve this problems? When I try to probe X ,y or z axis, the alarm 4 always on
[image]

kweilim98, 2022-06-15:
[image]
Some expert told me that is about the noise issue but I not sure where is

abinoer, 2022-06-15:
When you say probe x, do you mean the x axis goes back to zero? We do not currently support Probe XY.Also, does your host support the GRBL protocol?

kweilim98, 2022-06-15:
I am using the other software for control, no matter if I am probing the z, x, or y-axis, the alarm still triggers. This is the first time alarm 4 happen in probing action. Because before this when I am probing don't have these issues.

kweilim98, 2022-06-15:
i am using the CNC 3D commander software for my board control
[image]

kweilim98, 2022-06-15:
https://user-images.githubusercontent.com/66898187/173762077-3a038156-5713-4b32-91d8-8fd3a1094937.mp4

abinoer, 2022-06-15:
Try setting $6

kweilim98, 2022-06-15:
$6=1 already, if $6=0, the probe will be normally open

kweilim98, 2022-06-15:
when probing with the $6=0, the alligator clip and touch plate need to stick together, that is not the correct way for probe

abinoer, 2022-06-15:
Set $6=0, try to see if it works.

kweilim98, 2022-06-15:
it works, but in the wrong direction, when $6=1, zaxis probe moving z-, when $6=0, z-axis probe moving z+, means the wrong direction. And the alligator clip and touch plate need to stick together, if not it detect already done probing

kweilim98, 2022-06-15:
is it the program problem, motherboard problem, software problem, or my wiring problem? since I am using the gnd and signal only from the probe port

abinoer, 2022-06-15:
[image]
The command I use is: g21G91G38.2z-50f80; G0Z0. 5; G38.2 Z - 2 f30; G92Z10; G1Z10F80

kweilim98, 2022-06-15:
ok, nvrmind, i will trying using the candle software and reply to u later

kweilim98, 2022-07-04:
> [image] The command I use is: g21G91G38.2z-50f80; G0Z0. 5; G38.2 Z - 2 f30; G92Z10; G1Z10F80

I have try already with the candle software and the code that u given , still have the alarm 4 probe issue, what is the problem actually?

abinoer, 2022-07-05:
[image]
Is $6 already set to 1?

kweilim98, 2022-07-05:
> [image] Is $6 already set to 1?

Yes