makerbase-mks/MKS-DLC32 #237: Z Probe

Q8bilon, 2022-06-17:
Thank you for the new firmware ( V2.1.0_H35_20220510_N_CNC_001.bin ) which is better than previous firmware version.

After measuring the touch plate depth 13.5mm where should I write the probe command for the Z axis ? or How the probe works with the new firmware ?

I Have one more request Please, on the new firmware you set the movement as 0.1, 1.0, 10mm is it possible to add 100mm also? Thanks

Q8bilon, 2022-06-20:
https://user-images.githubusercontent.com/107723896/174547630-3c266588-76e8-45ad-bb8a-9380086d03f6.mp4
Could be like that ?

Q8bilon, 2022-06-21:
I'm So sorry but now I do erase a case against Aliexpress seller of MKS to refund my money back for two offline boards it's not working for CNC with a video showing the probing action and the movement only 1cm 10mm it's really ridiculous issues and they don't like to fix it ?   it's better to explain that in youtube So the new buyers will know.

abinoer, 2023-01-09:
Sorry, we don't have the energy to modify for every DIY customer, the firmware is already open source

closed 2023-01-09 by abinoer