makerbase-mks/MKS-DLC32 #239: Constant voltage at the pins of the laser connector

zmei46, 2022-06-22:
Hello!
Help fix the problem of constant voltage 11.4 v on the outputs of the laser connector. There is a constant voltage of 11.4 v on pins G and V which causes the laser to burn constantly at full power. At the TTL pin, the voltage appears as expected when the LazerGRBL program is running, but the laser is not controlled, it is constantly on.
MKS DLC32 board, firmware V2.0.8_H35_20220105_N.

TimE83, 2022-06-23:
I'm having a similar issue, though I have a 2 wire laser 12v 5.5w. and a breakout board for it. I'm also using a 3018 machine. But using the various types of firmware, seem to be getting 12v at both spindle and laser ttl 3 connection as you say. I can't find anywhere to change pin definitions. I believe it's up to the firmware.

zmei46, 2022-06-24:
I used the connection according to the attached instructions to the board.

[image]

abinoer, 2022-06-25:
@zmei46 The voltage of G and V is constant. There's no problem. Please confirm whether your laser is controlled by TTL signal, maybe there is something wrong with your laser.

zmei46, 2022-06-25:
The laser is fine - I tested it on my other machine with a different board and it works great. Everything on this board is adjustable and works great. I checked the signals for G and V - the voltage appears only when the burnout program is turned on.

[image]
[image]
.

abinoer, 2022-06-27:
Is your laser TTL signal controlled？Or through the 12V PWM control

abinoer, 2022-06-28:
[image]
OR
[image]
Try the preceding two connection modes

zmei46, 2022-07-03:
Today I made a connection according to the second scheme - the laser control worked, although the laser does not turn off completely, but continues to burn and the power can be increased with a regulator on the laser. Well, in general, you can work, but you have to adapt. I wish the programmers would fix this.
Сегодня сделал подключение по второй схеме - управление лазером заработало, правда лазер не выключается полностью а продолжает гореть и мощность можно увеличивать регулятором на лазере. Ну вообще работать можно, но надо приспособиться. Хотелось, чтобы программисты исправили это.