makerbase-mks/MKS-DLC32 #240: With New Firmware ( V2.1.0_H35_20220526_02_N_CNC ) The Probe Fail

Q8bilon, 2022-06-27:
After setup the touch plate depth on settings $46=(thickness setting) And $6=1 it gives Error massage ( Info  Probe Fail ) ,,,,After touch the plate the Z axis probe go down not up ?

abinoer, 2022-06-28:
Are you a simulation tester? It needs to make a second contact.

Q8bilon, 2022-06-28:
Second contact Down !!? After touching the probe plate for a second Stop, And go down again not up like 5mm or 10mm !? Try to see what new in setting also I found $38=1  $40=1  ? I don't know what it's stand for,, Blank ?  For the previous version firmware 001 the probe touch the plate and go up 10 mm but there is no $46 and $38 $40,,, With New firmware they fix the Spindle issues no need Spindle frequency that was a good move, But they did not fix also the option steps axis movement from 0.1 to 10mm I do need 100mm on X and Y for tuning CNC and easy movement. Thanks

JSF9026, 2024-11-23:
can you share that firmaware version file ?