makerbase-mks/MKS-DLC32 #241: Z movemenet buttons

Kabsyslab, 2022-07-02:
Hi,
with the latest firmware update for CNC there are no more Z movement buttons in the WebUI. This makes the webui useless!!!!
Please Fix this as soon as you can
thanks

Kabsyslab, 2022-07-02:
hi,
I think i know what went wrong, the webui is the one for the laser engraving not cnc. you must have mixed up the files when you compiled the up date

GraftSV, 2022-07-21:
Greetings
the same problem, after the update, the buttons on the Z axis disappeared
Firmware V2.1.0_H24_20220511_01_N_CNC.bin
Also, the Z-probe does not work properly.
and frequent freezes

tbalja, 2022-10-20:
I bought the MKS DLC32 and TS35 LCD board and it has CNC Board V-20 V2.1.0_H35_20220510_N_CNC_001 firmware installed. I have the Z axis adjustment buttons correctly displayed on the screen, but there are no Z axis buttons on the WEB UI. Likewise, there are no buttons on the WEB UI for adjusting the speed of the spindle motor, but for turning the laser on and off. This looks like the WEB UI is only intended for a laser cutter and not for CNC, although on the initial screen I have an indication that the firmware for CNC is installed.