makerbase-mks/MKS-DLC32 #243: problema minimo pwm

distebia, 2022-07-04:
salve, ho un problema con una mks dlc32 v2.1 e firmware 2.08 e 2.10 per laser normale board 2.0. il problema è che non riesco ad accendere il laser sotto il 10%. con macchina arduino montato nella stessa andavo a variare il parametro mandrino_pwm_min_value contenuto in config.he impostavo il valore di 25 per aggiungere mV all'uscita pwm, per avere la minima accensione.
i miei parametri su arduino uno sono $30=255, $31=0. Con arduino configura in questo modo accendo il laser al 1%. Con mks dlc32 v.2.1 non ho modo di fare la stessa cosa in quanto non ho il config.h. Ho trovato il parametro $28 che, momentaneamente, ho lasciato a 1000, $35 che dovrebbe essere l'equivalente di mandrino_pwm_min_value ma variandolo non ottengo differenze. Ho anche impostato $31=100 ma accendo sempre il laser intorno al 10% e, anche variando $30, $31, $35, $28 non ottengo variazioni di emissione pwm.
tutto questo sia con lasergrbl che con lightburn (LB configurato come da $30). è come se la scheda memorizza i valori dati ma non li esegue, o li esegue entro un certo limite.

distebia, 2022-07-05:
ho un co2 50w e ho montato arduino uno grbl 1.1h. nel pin pwm per raggiungere 0.2v e accendere il tubo al minimo dovevo commentare il valore del config.h   //spindle_pwm_min_value=25 per accendere. uso $30=255, $31=0. smonto arduino e monto la mks dlc32 v2.1.
imposto $30=1000, $31=100 (non posso commentare il config.h) e se invio g1m3s100f1000 il laser si accende. $28 lasciato a 1000 di default. invio g1m3s90f1000 il laser non si accende. in tutto questo se imposto il $35 con valori 10,20,30 non ottengo variazioni. tutto questo comporta che, in modalità m4, ho dei cerchi e quadrati incompleti in quanto si spegne quando rallenta.

distebia, 2022-07-05:
> Perché non puoi sparare laser al di sotto del 10%? Inviare l'istruzione "M3 S50 G1 F1000" non funziona? 50/1000 = 5%

non funziona, non accende. ho bisogno di alzare mV per l'uscita minima e raggiungere 0.2V al 1%

abinoer, 2022-07-06:
You said so much, but I still don't understand what you need, you set these parameters for what

distebia, 2022-07-06:
vorrei essere in grado di accendere il laser al 1%, attualmente lo accendo al 10%. inoltre se effettuo un taglio al 10% ho delle interruzioni in prossimità degli angoli o dove la macchina rallenta. se non correggo questo la macchina per me è inutilizzabile

distebia, 2022-07-07:
esiste un firmware per laser normale che mi da la possibilità di variare i parametri inerenti l'uscita pwm? grazie

distebia, 2022-07-09:
un conto è un firmware chiuso dove non posso cambiare indirizzi hardware o velocità processore ecc... un conto è impedisce dei settaggi basilari. da come ho capito tutti dobbiamo avere lo stesso alimentatore che si accende allo stesso modo, accetta la stessa frequenza di pwm. Se ho comprato il vostro prodotto completo di monitor lcd perchè devo mettere un altro firmware perdendo altri benefici?

distebia, 2022-07-13:
> Hai detto tanto, ma ancora non capisco di cosa hai bisogno, hai impostato questi parametri per cosa

esiste una soluzione al mio problema?

closed 2023-09-09 by distebia