makerbase-mks/MKS-DLC32 #244: Lasted CNC firmware issues

kweilim98, 2022-07-05:
Hi , here I found some issues about the lasted CNC firmware still have problems that need to fixed

1. Webui only has the features of laser user, not for the CNC user
IMG-20220704-WA0001

2. The buzzer features not on when
- hit the hard limit
- hit the soft limit
- driver enable while gcode was running
( The lasted firmware just add-in on-off buzzers function only)

3. When the driver was enabled, the gcode still keep streaming and non-pause (bug)

4. xyz probing not added ( so when the user using this kind of probing, it will come out with the alarm 4 probing fail issue)

5. every time when connecting the wifi with the password, it does not autosave (means every time when desire to connect need the key in the password

aviador58, 2022-07-14:
Hello, I want to use this board for my new CNC, what firmware is the latest one that you are using?

kweilim98, 2022-07-15:
> Hello, I want to use this board for my new CNC, what firmware is the latest one that you are using?

[image]

aviador58, 2022-07-15:
I don't see that photo.

aviador58, 2022-07-15:
you use mks dlc32 v2.1 for your CNC?

kweilim98, 2022-07-16:
Yes

kweilim98, 2022-07-16:
[image]

aviador58, 2022-07-19:
Thanks, I didn't know about this new update, I'm going to install it on my board and try it out.
A question that comes to mind, you have information or proof that this firmware can be loaded on an MKS-GEN_L, I say it because it has the possibility of having a 4 axis or even a 5 axis, very necessary for 3D machining.

kweilim98, 2022-07-19:
> Thanks, I didn't know about this new update, I'm going to install it on my board and try it out. A question that comes to mind, you have information or proof that this firmware can be loaded on an [...]

I not sure compatible for MKS gen L or not. But as on board show only able to support 3-axis only

JSF9026, 2024-11-23:
can you share the old cnc firmaware or the latest one ? the link is broken in the original thread