makerbase-mks/MKS-DLC32 #245: Is this board compatible with CNC milling at all?

mo3600, 2022-07-05:
Bought this board as an upgrade for my DLC V2.1, I flash the latest CNC firmware onto the board but cannot get it to work with grbrlController. Is this board compatible with CNC milling at all or is it just for laser engraving?

kweilim98, 2022-07-06:
> Bought this board as an upgrade for my DLC V2.1, I flash the latest CNC firmware onto the board but cannot get it to work with grbrlController. Is this board compatible with CNC milling at all or is it just for laser engraving?

What firmware u flash? CNC or laser?

mo3600, 2022-07-06:
Latest CNC firmware for TFT35

> > Bought this board as an upgrade for my DLC V2.1, I flash the latest CNC firmware onto the board but cannot get it to work with grbrlController. Is this board compatible with CNC milling at all or is it just for laser engraving?
> 
> What firmware u flash? CNC or laser?

Latest CNC firmware for TS35

mo3600, 2022-07-06:
> > Bought this board as an upgrade for my DLC V2.1, I flash the latest CNC firmware onto the board but cannot get it to work with grbrlController. Is this board compatible with CNC milling at all or is it just for laser engraving?
> 
> What firmware u flash? CNC or laser?

Latest CNC firmware for TS35

kweilim98, 2022-07-06:
> > > Bought this board as an upgrade for my DLC V2.1, I flash the latest CNC firmware onto the board but cannot get it to work with grbrlController. Is this board compatible with CNC milling at all or is it just for laser engraving?
> > 
> > 
> > What firmware u flash? CNC or laser?
> 
> Latest CNC firmware for TS35

What controller you use? I ok with that board

mo3600, 2022-07-06:
> at controller you use? I ok with that board

I'm trying to use grblController on my computer as a controller. Do you use this board for CNC milling with grblController?

kweilim98, 2022-07-06:
> > at controller you use? I ok with that board
> 
> I'm trying to use grblController on my computer as a controller. Do you use this board for CNC milling with grblController?

Yaa, I am using a lot for testing, so far the ugs, candle, cncjs, quite ok

Wiki-the-Tinkerer, 2023-09-03:
Only sender I have got it to work with for somewhat is OpenCNCPilot. But the board looses connection all the time if the motor (spindle) is on, regardless of sender. Different senders then handle this in different ways, sometimes making it look like its the sender, not the board that f:s up. 
I have hardwired the spindle directly, with no connection to the board, and even tried connecting it using a 50m cable, from a completely separate circuit, just because I got stubborn. Endstops are octocoupled. But if the spindle is on, the board looses usb connection (short hi quality USB wire - longers dont work at all, even with booster) connection all the time. Endstops are octocoupled. No chance of getting even a simple job done.