makerbase-mks/MKS-DLC32 #246: New Laser V2.10 Firmware: Bugfixes not at all?

Lasermax3000, 2022-07-07:
Hello MKS,

after long time of no improvements, i was really happy to see a new update of the Laser Firmware has been released.
But i have to say, i am more than disapointed from the new V2.10. However MKS doesn´t want the help from the Community, by making the Source Code open. 
So here is a short List of Bugs i´ve found so far:

1.The XY Coordinates are not shown anymore at the Mainscreen, they are always 0.
2. The Homing Process is not working at all anymore, likewise Softlimits do -> Short Video will be uploaded soon. 
3. Framing is sometimes not working, you get kicked out to the Home Menu.
4. Uncontrolled Motions after Pushing some of the Manual drive buttons (10mm fast), it even drove in the wrong direction for a short time!?

Also a few more questions:
Which Parameter has to be changed to adjust the Speed for Patrolling/Framing an gcode File?
What is with continous Jogging or if its not possible maybe a 5cm Step Button?
Why is it still not possible to set a Fixed IP Adress request in the settings, on other MKS Boards itˋs part of the Firmware, so it has to be possible?
Is there any way to Control Lightburn over Wifi?

I would be happy to read something of your future Plans to this Board.
Summary: I will go back to V2.08 till these issues will be fixed, because its more useless than it was before, and more dangerous. -> MKS embarrassing to release something like this, did you even try it?

https://user-images.githubusercontent.com/99832588/177883921-934a15e1-2f6d-465f-a12f-566e07f21414.mp4

https://user-images.githubusercontent.com/99832588/177884136-ef03d3f1-0a95-4bdd-8c16-0fcaec22064d.mp4

Lasermax3000, 2022-07-08:
Here two short Clips to show the Homing issue.

https://user-images.githubusercontent.com/99832588/177936465-789623b7-a3bc-47fa-8b60-76a7920d15d0.mp4

https://user-images.githubusercontent.com/99832588/177936509-f73d3d6f-e93b-4886-bd1f-80ad5b099fb3.mp4

Insigma1, 2022-07-09:
I've also reverted to V2.0.8_H35_20220105_N, having found X axis to be reversed when controlled from the LCD only.

abinoer, 2023-03-02:
Sorry, limited energy, not timely reply. The firmware has been updated, please try burning the latest firmware

closed 2023-03-02 by abinoer