makerbase-mks/MKS-DLC32 #248: Pinmap

RVAMike, 2022-07-08:
I wouldn’t really call this an issue I’m looking for the pinmap Showing what pins Numbers and Functions are on the  DLC32 V2.1 trying to get it configured with FluidNC thanks for your help

manitoulin-maker, 2022-07-12:
wont ever be released

abinoer, 2022-07-19:
[image]
https://github.com/makerbase-mks/MKS-DLC32/tree/main/hardware/MKS%20DLC32%20V2.1_003

thiagolocatelli, 2022-07-22:
Those PDF's are not very helpful to those who can't read schematics and the PIN.pdf is incomplete. Can you add a Pin Map table to the README.md file the same way it was done for the  TinyBee? Example: https://github.com/makerbase-mks/MKS-TinyBee#pinmap

abinoer, 2022-08-10:
The IO on TinyBee's PinMap is also extended by the extension chip, which is not available without the schematic

closed 2022-08-10 by abinoer

errolt, 2023-01-10:
And the pin numbers in the PIN MAP pdf is wrong. 
SD DO: IO2 is actually IO12
EXP2 IO39 is actually IO19