makerbase-mks/MKS-DLC32 #251: V2.10_H35_2022_0621_N_ZX_001 X-axis problems

antude1, 2022-07-10:
The X-axis is inverted, you have to change the icons on the screen that make the X-axis move. 
The laserGBRL connection is correct and I wasted all day trying to adjust the X axis, inverting the motor cables, the motor direction values, the homing values and I always had the same problem. The homing is correct but when I press the X axis direction keys the motor goes in the opposite direction.

It is necessary, that they revise the firmware and correct this error.

Translated with www.DeepL.com/Translator (free version)

quattrohaje, 2022-07-17:
Hello, i have same problem. another problem is that my webgui stopped working and I can't set, for example, the frequency of $28 . Otherwise, your HW and FW are wonderful, just to tighten a few details. 
Vlada

wfvn, 2022-08-23:
Same here. X is reversed. First run I configured the laser according to the touchscreen (which works!!!) But then Lightburn was unable to wiel untill i reversed x the wrong way

abinoer, 2023-01-07:
It has been resolved, please update the firmware

closed 2023-01-07 by abinoer