makerbase-mks/MKS-DLC32 #253: SD Card usage causes reset of DLC32 v2.x

keithstrachan, 2022-07-15:
Hi, I am having issues with my DLC32 with various firmware, when a SD card is inserted and i am using SD for Laser cut/engrave - board resets at random times.
Firmware tried -
V2.0.8_H35_20211222_N.bin
V2.0.8_H35_20220105_N.bin
V2.10_H35_2022_0621_N_ZX_001.bin

SD Card tried -
8Gb Class10
16Gb Class10
32Gb Class10

i see this is also raised before and closed #79

prosoftJit, 2022-10-29:
Im having this same problem with lastest firmware

abinoer, 2023-03-02:
Sorry, limited energy, not timely reply.

We have never had such a problem. Maybe your card caused the short circuit. We hope you can upload the video so that we can find out the cause. thank you