makerbase-mks/MKS-DLC32 #256: Web Interface Fails

baylockzathras, 2022-07-19:
Hi All, Running:
V2.10_H35_2022_0621_N_ZX_001

Seems when the Device accesses the SD Card, the web interface stops working.

Steps to re-produce:
Turn on with SD Card inserted.
or
Use LCD to show boundary of Burn.

Currently working around by:
Power cycling with SD Removed, then inserting SD Card after.

Has any one else found this issue?
Have a better solution?

Thanks.

abinoer, 2023-03-02:
Sorry, limited energy, not timely reply.

We have not received similar feedback, maybe you can burn the latest firmware to see if the problem you mentioned still exists. If there is still this problem, we hope you can upload the video to see, so that we can find out the reason, thank you

baylockzathras, 2023-03-03:
Apologies, I forgot about this post. Seems to be a compatibility issue with chrome. Works fine in edge.

closed 2023-03-06 by abinoer