makerbase-mks/MKS-DLC32 #257: wrong dimensions in the laser engraving in my acro system

l0v3byte, 2022-07-20:
Hi,
I installed this card in my acro system with TMC 2225 and set the parameters to perfection and installed the latest version of the firmware and everything works correctly but when it affects the dimensions they are really small in a 230mmx180mm logo it came as small as a postage stamp either with offline controller than with LaserGRBL and Lightburn. I tried to reinstall the firmware and put the correct parameters back but nothing. This problem occurs only in the engravings if the axes are sold manually spotted both by the software described and by the offline controller there are no problems they are moved correctly for the entire useful work area.
Can someone help me.

abinoer, 2022-07-25:
Is it the right distance? You can set the dial to adjust the subdivision by setting $100-102 to adjust the pulse

l0v3byte, 2022-08-07:
Hi @abinoer for the travel resolution on Y and X is settinig to 228.571.
the machine use the belt gt2 large 5mm and 2mm of pinch and 14 teeth pulley and im using TMC 2225 for stepper driver