makerbase-mks/MKS-DLC32 #258: I the source code available for this?

growlybearproductions, 2022-07-21:
I looked around but could not find it if it is.

Medina-ui, 2022-07-21:
There are several requests for it, and MKS apparently is not willing to open 
I would very happy if they release it

troth530, 2022-07-25:
MKS is in violation of GPL.  People who purchase their hardware should consider that overall they are harming the community.  This design is a very poor design, very slow in processing gCode anyway.

ubergeekseven, 2022-08-27:
They basically stole the esp32 grbl project and modified it. They need to publish their source. The esp32 grbl people are fine with it as long as they state it.

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2022-11-15 by makerbase-mks

troth530, 2022-11-15:
@makerbase-mks thank you.  I am interested in integrating your touch screen.