makerbase-mks/MKS-DLC32 #260: autosquare ponte Y

distebia, 2022-07-24:
Salve, ho una mks dlc32 e desideravo sapere come realizzare un sistema autoquadrante del ponte. so che devo utilizzare 2 finecorsa e 2 motori indipendenti per l'asse Y ma non so come collegare i 2 motori e i 2 fimecorsa. a bordo della mks non ho montato i driver ma utilizzo 4 tb6600 (1 per X, 2 per Y sdoppiando i segnali e 1 per Z) comandati dall'uscita esterna per driver esterni, (E, S, D, G). per i finecorsa forse il secondo dell'asse y viene condiviso con Z?.

abinoer, 2022-07-25:
[image]
What instruction are you using? I use Candle. I use this command

abinoer, 2023-01-09:
Salve, ho una mks dlc32 e desideravo sapere come realizzare un sistema autoquadrante del ponte. so che devo utilizzare 2 finecorsa e 2 motori indipendenti per l'asse Y ma non so come collegare i 2 motori e i 2 fimecorsa. a bordo della mks non ho montato i driver ma utilizzo 4 tb6600 (1 per X, 2 per Y sdoppiando i segnali e 1 per Z) comandati dall'uscita esterna per driver esterni, (E, S, D, G). per i finecorsa forse il secondo dell'asse y viene condiviso con Z?.

https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE/issues

abinoer, 2023-01-09:
Sorry, we don't have the energy to modify the custom requirements for each DIY customer, the firmware has been open source, you can modify yourself

https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2023-01-09 by abinoer