makerbase-mks/MKS-DLC32 #262: mks dlc32

benoit251, 2022-07-27:
bonjour, j'ai achete votre carte mere mkl dlc32 v2.1, que j'utilise avec un laser, pour piloter le tout j'utilise lightburne, malheureusement a chaque fois que je lance une gravure laser, j'ai le message d'erreur suivant qui s'affiche:

[MSG:M4 requires laser mode or a reversable spindle]
error:20
Commande g-code non prise en charge ou non valide trouvée dans le bloc.
En ligne ou quasi en ligne 4 :
Travail interrompu

si je me met en M3 le laser bouge mais ne s'allume pas, alors que en M4 le laser s'alume commence a graver sur 1 cm s'arrete , continue a bruler sans bouger et m'affiche le message citer au dessus

pourriez vous m'eclairer sur cette erreur

Auparavent j'utilisais la plaque MKS dlc v2.1, et je n'avais aucun probleme avec cette derniere

en vous remerciant d'avance

cordialement Benoit Detours

EU4AX, 2022-07-28:
I have absolutely the same problem. 
If i correct understanding, here []https://github.com/bdring/Grbl_Esp32/issues/946 we could have a solution for resolve it self. But Makerbase laid in firmware codes for launching nuclear warheads, so the source are closed

EU4AX, 2022-07-28:
[benoit251](https://github.com/benoit251), please rename thread if this possible. Add something like an "M4 error in laser mode"

benoit251, 2022-07-28:
bonjour,

j'ai resolu le probleme en telechargeant et en installant le dernier firmwar fourni sur un autre poste par un internaute, le dernier firmawar installe grbl 1.1h qui resaut le probleme

ci-dessou le lien ci-dessous avec les firmwar et les boite a outis

https://github.com/makerbase-mks/MKS-DLC32/archive/refs/heads/main.zip

EU4AX, 2022-07-28:
[benoit251](https://github.com/benoit251), tnx! 
V2.0.8_H24_20220107_N.bin working properley, although when board connected in console i get 1.1 without a letter.

alstare72, 2022-11-19:
> Benoit Détours

Est ce que vous pouvez postez une photo de votre installation du mks carte .  Car problème allumage laser brûle tous le temps .. merci