makerbase-mks/MKS-DLC32 #264: Wish for a new laser firmware

EU4AX, 2022-07-28:
Hi, MKS!
Take into account my whim, please.
I would like in the new firmware in the menu CONTROL->LASER to have a button 1% power instead 50%. Or even 0,5%. This would be useful for manual focusing of the lens. 
Dont forget about users who have 2.4 TFT, please
[image]

It would be great, of course, to be able to configure this parameter yourself.  
Perhaps this can be implemented in MKSLaserTool in the new section. It would also be convenient to be able to edite and add user macros to the ESP32-Web.

Thanks

danielinkiel, 2022-08-07:
any news? solutions? ideas?

Greetings from Kiel

EU4AX, 2022-08-08:
In addition to the already requested?
Well, you can lay out the source ;)