makerbase-mks/MKS-DLC32 #265: Isnt somebody smart enough?

manitoulin-maker, 2022-08-03:
HERES A GREAT IDEA EVERYBODY! 

Why dont we all try to decipher the board since MKS wont. there is no reason for them to not release the pins for this board. we are all loyal to them even tho they wont release it. im not sure if i would call it "false promises" , because im not sure they ever said they would, regardless, they shouldnt have to, this is all "OPEN-SOURCE" so i think they should really rethink their place in this market.  

ANYWAYS !!!, ....  I see the half-a$$ pinout they have so far. i count a good amount of IO Pins that are listed, maybe we can figure the rest out with some reasoning, somewhat like what jailbreaking apple, im not saying lets exploit ANYTHING, what i am saying, is, since it is open source, cant we try to work our way through the firewalls persay. 

    Heres my idea:        

                        from the given IO pins, we can work out what ins are left, and they try to work out the logical places for those IO pins from either like chipped boards, or just deductive reasoning, now i dont want anybody to fry thier boards, so we should limit pin checking to the low voltage ones from cross checking the pinout of the ESP chip.        Just an idea, as i am getting EXTREMELY TIRED OF MKS'S BULLSH#T.  

  I am okay at compiling, adding some new features but really i am still at a user level with configuring marlin, in no way do i understand how c++ works or anything other than you need defines and .h files thats IT. so, i/we need all of your help, please. we all can do this!! look at what other groups do when they come together, we are all makers! please for the sake of the human race, lets figure this out ugh!!!  i wanna make sh#t with my 9 yr old son for the love of frig.

manitoulin-maker, 2022-08-03:
Found out some more pins last few hours, i will post my findings tomorrow. i have 4 of these boards SO ILL TRY TO BE AS MUCH OF A HELP AS I CAN,

Lasermax3000, 2022-08-07:
I would also like to help, but actually i don´t have the time and the opportunity to measure which Pins from the ESP32 go out to which Plugs. But if anyone can figure it out, i can help to develop a simpler Opensource Firmware which works better than this current BULLSH#T.

manitoulin-maker, 2022-08-07:
That's all we need. Anybody that can even help a tiny tiny bit is all we
need to do this
 Maybe if we get close. They will just release it.  Through looking at some
other firmware for the dlc32. There's about 20 pins defined already.
Fluidnc is where I. Getting a lot of I do as they have compiled for that.
We will get it guys.

keithstrachan, 2022-08-09:
the main issue is the LCD screen. The functional pins are already working in other firmware, like GRBL-Hal and FluidNC, but no one has the LCD working... this is the main problem.

kweilim98, 2022-08-10:
> the main issue is the LCD screen. The functional pins are already working in other firmware, like GRBL-Hal and FluidNC, but no one has the LCD working... this is the main problem.

yaaa, agree with u

keithstrachan, 2022-08-10:
i think that what is needed to start, is to
1. find an example to use LCD with ESP32 chip, and see how that related to io pin usage.
2. append the grblhal code into the lcd screen code to create the new platform.

i believe that is why it is hard, because people are looking at the grbl side first, and i think that it needs to look at the lcd side first.
i'm not a programmer, but i think this way logically works

keithstrachan, 2022-08-10:
touch appears to be based on TSC2046 / ADS7846 touch screen controller. not sure if the xpt2046 library will drive this, but i believe they're the same family.
probably using the TFT_eSPI driver to run the screen

Medina-ui, 2022-08-10:
Hi. I want to help on this because I have several boards and tft's are arriving soon ( but I don't have a lot of free time ) . About the screen, I am not sure because TFT's are still on the way, but maybe this is  not a big problem.... Check this board (MKS TinyBee) , same ESP32-WROOM-32U and TFT possibilities (MKS35) with two 10pin conections cables, and that is clear for me that all logos,bitmaps and "letters" are on Marlin/GRBL side, just like Arduino with LCD12864. And libraries are available for this on the link:  https://github.com/makerbase-mks/MKS-TinyBee/tree/main/firmware

keithstrachan, 2022-08-11:
> Hi. I want to help on this because I have several boards and tft's are arriving soon ( but I don't have a lot of free time ) . About the screen, I am not sure because TFT's are still on the way, but [...]

Hi Medina,
There are many examples of using the standard mks tft35 screens because they run their own processor, stm32f4 etc.
This is easy to get going.

The TFT screen supplied with the MKS-DLC32 is a MKS TS35-R which does not have it's own firmware or processor, but uses a driver and is controlled by the DLC32 ESP chip.  Similar design to the TS35-E3, TS35 and PI-TS35 - these all use SPI to control screen and touch.

the MKS TFT/TS repo is here: https://github.com/makerbase-mks/MKS-TFT-Hardware
you can look at the schematics to see what i mean

Medina-ui, 2022-08-11:
Hi

> Hi Medina, There are many examples of using the standard mks tft35 screens because they run their own processor, stm32f4 etc. This is easy to get going.
>
> The TFT screen supplied with the MKS-DLC32 [...]

HI Keithstrachan, I understand exactly what you mean. I think you did not understand me, but no worries, as soon as my TFT35 R arrives I will test "my theory" and share.

giordy72, 2022-08-21:
Look at the schematics for pins and connections. 
Switch to fluidnc or gribl Hal, and forget the unusefull tft. Use an old smartphone as hmi  and your up to go in half an hour.

keithstrachan, 2022-08-21:
thank you for your unusefull comment giordy72, so small minded, can't see the bigger picture

Medina-ui, 2022-08-30:
Hi guys, I had a look on MKS documents and compare V1.1_0001 SCH and PIN Documents are there are a lot of mess on pins description/assignment . Also I had a look on V2.1_003 Documents and this one have only one pin that is diferent on both documents. In the file you can see in red the diferent information on both files for the same board, and on orange a strange assignment. Does MKS want to give us fake information?
[image]

keithstrachan, 2022-08-30:
> Hi guys, I had a look on MKS documents and compare V1.1_0001 SCH and PIN Documents are there are a lot of mess on pins description/assignment . Also I had a look on V2.1_003 Documents and this one [...]

@medina-ui its not fake information when there is a typo in the document. I have been over these as well, and there are mistakes, but i believe they are honest accidental mistakes, not deliberate. Deliberate mistake (fake information) is if they put information that is contrary to a truth, making a lie.
You must also remember that for many people, english is not first language, and when you are typing on a keyboard in a hurry, sometimes you forget to type a letter, number, symbol.. just like in your post, you spell "different" incorrectly :)

So far, the only thing wrong is that MKS indicated that they would release the source code, and they have not.
during any PCB development, just like code development, the designer/writer will make small changes to optimise the artefact. this is why they have v1, v2 etc.  When you compare v1.1 of a design against v2.1 - you should expect variations

But also keep this in mind. Any code, grbl, grbl-hal, fluidNC, marlin etc - have config files to allow the specification of the pins to allow tweaking to align (fix up mistakes) in the physical board.
if the software was completely hard coded, then we would be in real big trouble.

kweilim98, 2022-08-31:
https://github.com/solawc/ESP32_LaserBox_E/blob/main/GRBL/src/Machines/i2s_out_corexy_mks_dlc32.h

I not sure here the Pin out able to help u or not because I am not familiar to coding, but here is about the MKS dlc32 xyz pin out, u can check it out at here 

[image]

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2022-11-15 by makerbase-mks