makerbase-mks/MKS-DLC32 #268: X und y Achse laufen gleichzeitig

Experiment73, 2022-08-12:
Moin, ich brauche Hilfe. Ich nutze einen einfachen Diodenlaser mit 2 y Achsen Motoren und 1em x Achsen Motor, Z nicht vorhanden. Als Programm Lightburn. Da ich ein Board wollte mit Offlinesteuerung habe ich mir nun für mein gerät das MKS-DLC32 V2.0 gekauft. Leider funktioniert da einiges nicht so wie ich dachte.  Mein Problem ist, dass wenn ich die x Achse oder die y Achse bewege immer beide Achsen angesteuert werden. also als Beispiel x+10cm und ausgeführt wird dann vom Controller sowohl x als auch y +10cm. angezeigt wird nur x+10.  wie und wo kann ich dieses Problem beheben? Im Moment kann ich dadurch das board nicht nutzen da dadurch gravieren mit dem mks board unmöglich ist. Sollte das Problem schon irgendwo besprochen worden sein, hab Ichs leider nicht gefunden. Dank für die Hilfe im Voraus

hans-de, 2022-08-23:
Das Problem scheint nur die cnc Version der Software zu haben. Die aktuelle Laser Version hat das nicht. Dafür gibts andere Probleme.

Plottmania, 2022-09-01:
Nun, eventuell die Zeile $10=0 einstellen
Ein Cycle für Homing:

Abfrage einer Referenzfahrt:

$Homing/Cycle0=XY Der erste Zyklus XY wird zur gleichen Zeit referenziert
$Homing/Cycle0=X zur Referenzfahrt X im ersten Zyklus, $Homing/Cycle1=Y zur Referenzfahrt Y im zweiten Zyklus.
$Homing/Cycle1=Y Homing Y im zweiten Zyklus.

Ist Dein DLC32 richtig eingestellt, Dein LightBurn auch kannst Du machen was Du willst aber würdest nicht dieses Problem haben.

spexx73, 2022-10-29:
> Moin, ich brauche Hilfe. Ich nutze einen einfachen Diodenlaser mit 2 y Achsen Motoren und 1em x Achsen Motor, Z nicht vorhanden. Als Programm Lightburn. Da ich ein Board wollte mit Offlinesteuerung [...]

Habe das gleiche Problem. Anscheinend kann man jetzt nirgendwo mehr überhaupt eine Firmware herunterladen ;-(

Plottmania, 2022-10-29:
Warum nicht?
Es ist klar wenn Homing dedrückt wird beide Achsen verfahren?
Es ist klar wenn nach dem Homing oben rechts angezeigt wird X= 3 (Beispiel) und Y=3 (Beispiel)?
Die 3 mm sind dann in Deinen Settings eingestellt (freifahren der Endschalter)
Da ist jetzt Dein Offset und kann man einstellen...

https://wiki.the-iskens.com/

Was die DLC  betrifft könnt Ihr hier gucken:

https://plottmania.de/article/38-makerbase-dlc-familie/

Spaß haben