makerbase-mks/MKS-DLC32 #270: frimware Laser V2.10_H35_2022_0621_N_ZX_001.bin

Wieslaw68, 2022-08-22:
V2.10_H35_2022_0621_N_ZX_001.bin Laser. Invalid x + / x- direction in control.

hans-de, 2022-08-23:
Same problem here. The x direction between display and PC control is reversed. No Idea which of them is right. I hope it will be fixed soon. However, it would be better to release the source, the community can do this faster.

hans-de, 2022-08-25:
After a week of fruitless attempts with several firmware versions to bring the board into a usable configuration, I give up. The board goes back. Too bad, because the hardware itself is very usable.
I think this was my first and also my last MKS Board. 
With every new Firmware version they fixed some bugs and added new one.

Wieslaw68, 2022-08-25:
I don't know why they don't want to share resources, the community would have fixed it long ago.

tbsmnea, 2022-08-26:
The hardware as such is ok. The touch display also makes a good impression, but unfortunately there is no option to rotate the display by 180 °. Unfortunately, the software is full of bugs and MKS is probably not interested in resolving it. The decision for the board a lot for me to improve the good price, the super fast CPU and the WLAN support. Unfortunately, the programmers don't even get distinguished right and left. The X directors are exchanged via the display control. The over 70 bugs, which have also been reported on GitHub for months, speak a clear language here. I think the community could fix the bugs in a short time, but unfortunately the souce code is not accessible. For me, this is probably the first and last MKS board and if I don't get the board to run at short notice, it goes back. Unfortunately, I wasn't aware of that before. It's a shame about time.

iherbak, 2022-09-08:
I uploaded the firmware amd set up wifi, but the web ui is unable to load up. Any solutions? I had to revert the previous one to get it work

iherbak, 2022-09-10:
> Same problem here. The x direction between display and PC control is reversed. No Idea which of them is right. I hope it will be fixed soon. However, it would be better to release the source, the community can do this faster.

You have to edit the settings that's all, here is the default:

$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=2 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=500.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=80.000 (X-axis travel resolution)
$101=80.000 (Y-axis travel resolution)
$102=250.000 (Z-axis travel resolution)
$110=2000.000 (X-axis maximum rate)
$111=2000.000 (Y-axis maximum rate)
$112=2000.000 (Z-axis maximum rate)
$120=6000.000 (X-axis acceleration)
$121=6000.000 (Y-axis acceleration)
$122=10.000 (Z-axis acceleration)
$130=200.000 (X-axis maximum travel)
$131=200.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

iherbak, 2022-09-10:
> For me it was the first time, not working either. I downloaded the web Ui, just an empty website, SD card, and the web UI, newly loaded, and see it has worked. Apparently there is also a bug that does not work with the SD card for the first time. After that I didn't have to remove the SD card anymore

without sd it fires up. But as soon as i put sd in it kills itself. I reverted back to the previous version.

capi92, 2022-09-18:
> V2.10_H35_2022_0621_N_ZX_001.bin Laser. Invalid x + / x- direction in control.

Same here

abinoer, 2023-01-07:
It has been resolved, please update the firmware

closed 2023-01-07 by abinoer