makerbase-mks/MKS-DLC32 #273: Is it possible to change the Framing speed?

linx1clarke, 2022-08-25:
Hi,  

I have my board set up and working(after 4 hours of troubleshooting), and I am happy with it for the most part, but is there a way to increase the speed at which the engraver frames the project(before you start it so you can position the piece of material)?

I find the preset speed very slow, and for bigger projects, when you need to frame it multiple times, it is taking a very long time!

Can anyone help with this issue-Can someone tell me how to speed up frame speed?

Thanks!

a76664289, 2022-08-27:
根据机型性能修改一下参数
最大速度设置
$100=5000
$111=5000
$112=5000

加速度设置
$120=1000
$121=1000
$122=1000
###

linx1clarke, 2022-08-27:
I see these are for the speed settings, but my speed settings are at 6000mm/min and it is not doing that.  However, my acceleration is at 500, but I'm not sure that solves the issue, because I am not talking about the engraving process, just the speed of the frame process to position the piece of wood.

abinoer, 2022-08-29:
Sorry, this speed does not support the user's own modification

serial-tintypist, 2022-08-29:
> Sorry, this speed does not support the user's own modification

Let me correct you - sorry, this whole firmware does not support user's own modification (because we have stolen an open source project and could not care less).

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2023-01-07 by abinoer