makerbase-mks/MKS-DLC32 #274: Unfortunately, the programmers don't even get to fix the over 70 bugs reported on GitHub for months. I think the community could fix the mistakes in a short time, but unfortunately the SOUCE code is not accessible

tbsmnea, 2022-08-26:
**The hardware as such is ok. The touch display also makes a good impression, but unfortunately there is no option to rotate the display by 180 °. Unfortunately, the software is full of bugs and MKS is probably not interested in resolving it. The decision for the board a lot for me to improve the good price, the super fast CPU and the WLAN support. Unfortunately, the programmers don't even get distinguished right and left. The X directors are exchanged via the display control. The over 70 bugs, which have also been reported on GitHub for months, speak a clear language here. **I think the community could fix the bugs in a short time, but unfortunately the souce code is not accessible**. For me, this is probably the first and last MKS board and if I don't get the board to run at short notice, it goes back. Unfortunately, I wasn't aware of that before. It's a shame about time.**

EU4AX, 2022-08-27:
Since the main part of the boards is bought on platforms such as Aliexpress, it is worth stopping lazy and describing the real state of affairs with technical support in the reviews to the product.

Insigma1, 2022-08-28:
Also totally frustrated with MKS now , its clear they do not care about their products working properly, no updates for 2 months now latest firmware filled with issues , no source code released, I even contacted them directly on facebook only to be fobbed off, I will not buy their products in future nor will i recommend them to people.

EU4AX, 2022-08-28:
All these letters have no sense. The situation will change then and if a "professional" responsible for the firmware will be replaced by a person who will understand what and how works in this area. That the hidden sources do not guarantee the preservation of the workplace, especially when the requests to fix the bugs are simply ignored.
IMHO irritation of the popular YouTube blogger will have a much better effect than seething here, and writing letters in all kinds of reddits and facebooks

kweilim98, 2022-08-29:
https://github.com/solawc/ESP32_LaserBox_E

here I found the new programmer who now is overwriting the coding for mks dlc32, Is anyone who can help him to finish up the process of the new release of firmware will be