makerbase-mks/MKS-DLC32 #275: MKS DLC32 V2.1, flash read err, 1000, after new firmware

Dettinger-15, 2022-08-26:
Hello, I bought a MKS DLC32 V 2.1.
Then I have flasht a new firmware on this board
(V2.0.8_H35_20211222_n.bin).
Since then, the error has been under LaserGrbl leafls: "flash read err, 1000" and the connection is interrupted after a while.
I installed the firmware with the MKS laser tool and with the flash download tool version 3.6.6 from Espressif. Both lead to the same mistake.
Can someone help me with my problem?

abinoer, 2022-08-29:
Could you please provide a screenshot

Dettinger-15, 2022-08-29:
I bought this board
[image]
These are the error messages in LaserGRBL and about Serieller Monitor
[image]
[image]

hans-de, 2022-08-29:
Take Care that you have a complete Image. Its 4GB on size. I had the First time a similar Problem because the downloaded Image was broken

Dettinger-15, 2022-08-29:
I have take this firmware from the MKS Githup firmware folder to flash it on the board with the ESP32-Chip:
V2.0.8_H35_20211222_n.bin
it is has only 132kB.

hans-de, 2022-08-29:
Thats the Problem.Click on the Firmware to Open the folder and Download it from there. If you do a right click its corrupted.I did the Same mistake.

Dettinger-15, 2022-08-29:
Hallo hans-de,
Thank you very much, it is clear that the firmware update cannot work with the 132kB file.

closed 2022-08-29 by Dettinger-15