makerbase-mks/MKS-DLC32 #277: pen plotter attachment

kdaffy, 2022-09-03:
Hi. How could I use the MKS-DLC32 to control a servo to use it as a pen plotter?

renebohne, 2022-10-20:
I use FluidNC Firmware with the board and attached a servo to the z-axis. G0 Z0 and G0Z-5 are the g-codes for controlling the z-coordinates which control the servo... 
see: https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup#rc-servo

istinnstudio, 2022-11-14:
@renebohne  
Can you post more details on this, like a simple how to? config files? Is this the default FluidNC Firmware or a recompiled with modified code?

 additional useful links on this:
https://github.com/lavolpecheprogramma/grbl-1-1h-servo
https://www.instructables.com/How-to-Control-a-Servo-Using-GRBL/

helLf1nGer, 2024-03-18:
I used RC Servo 1501 MG. I connected to the LCD pwm pin (io25, 5v, gnd here is the schematics http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32). The travel is dictated from my config from z-5 to z5.
Here is my configuration for the motor (I need only 30°, so I mapped only from 1300 to 1700):
`  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 5
    homing:
      cycle: 0
      mpos_mm: 0
      positive_direction: true

    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.25
        min_pulse_us: 1300
        max_pulse_us: 1700

    motor1:
      null_motor:`

istinnstudio, 2024-03-19:
thanks... useful info