makerbase-mks/MKS-DLC32 #282: Laser module cooling fan always on

drkline, 2022-09-12:
Hello,
I have installed DLC32 boards on two laser machines and both of them have their laser module cooling fans running at 100% always.  The laser modules are Sculpfun S9 and S10.  I have had them both installed on a Ortur LaserMaster 2 and its control board and never had this issue.  What do I need to do to have the cooling fan run only when the laser is turned on?  Leaving it like this will eventually wear out the small DC fan on the laser module and I don't want this to happen.  Please let me know what I need to do.

Thank you,
Dave Kline

Plottmania, 2022-09-14:
This is normal and also good!
Unfortunately, mostly too thin cable collapses the performance.
Either a thicker cable and optionally a really good fan on it and neither the driver nor the LED get warm.
Really good it is with TEC and Peltier and we have about 22 degrees which helps a premature death or aging of the LED.
S10 with Peltier under full load 18 hours run without problems and no performance drop.
Sorry for my bad English

drkline, 2022-09-14:
Thank you Stephan,

So are you suggesting to have a cooling fan for the stepper drivers to keep them cool?  And what gauge wire would you suggest using for the laser module power?

Sincerely,
Dave

Plottmania, 2022-09-15:
Sculpfun S10 - Peltier-Cooling and correct cables

[image]

After a runtime of 18 hours under full load with 100% power measured: 10.2 watts optical output power.
The S10 from Sculpfun is a nice toy!

onehans1, 2022-10-18:
Maybe is good for you but for some not 🙉
Question stands. How to stop it from running at full whack non stop?
Original Atomstack board do not run when not needed.
Mks DLC32 v2.1 as soon i power it it goes crazy for no reason

hans-de, 2022-10-19:
Based in my knowing this ist not possible. But at least a cool laser is better than a hot. To make it save you musst also measure the temperature.

onehans1, 2022-10-19:
$S+ or $S pops up option where ~"TurnOnFanWhen LaserRunning=off" changing it to "on" doesn't affect state of fan

abinoer, 2023-01-07:
@drkline 
The motherboard output of 12/24V is a constant output, uncontrollable, this board can not do laser open to open the cooling fan

closed 2023-01-07 by abinoer

onehans1, 2023-01-07:
I'm talking about laser output pins S V G
Laser fan runs at full speed.
Atomstack frame A5 used fan speed according to laser power output strength