makerbase-mks/MKS-DLC32 #283: TMC 2160 driver -> dlc32

vfear, 2022-09-14:
Please, give the microstep jumper switching scheme on dlc32 for this external driver.  
I connect it via MKS CD 57/86.
[image]
[image]

abinoer, 2023-03-02:
[image]

closed 2023-03-02 by abinoer

LeSpectre25, 2025-03-28:
> hahaha...Barton Dring send my nice Adapters
> 
> [image]
> 
> Why? Because on the Y-axis two motors have to share one driver..so it's almost plug n play and I just have both signals at the outputs of the Y-axis and it runs really smooth, clean, quiet and without problems.

Good day, Could you please tell me what adapters are being used on the DLC board in place of the stepper drivers?