makerbase-mks/MKS-DLC32 #284: tmc2209 sensorless home

ahmetbozaci, 2022-09-15:
**Is it possible for sensorless home to work with mksdlc32?**

- Before I made sensorless home to mks robin nanov1.2. I solder the cable to the pin (next to enable pin- you can see in the picture) and connect it to the stop pin. It works well. For UART ı solder pins like in the [video](https://www.youtube.com/watch?v=7ShcFKXrVHo&fbclid=IwAR3jsRHUicc6HhGeMAGNvSYPr0pGor6llZIcXI9V2BUng4UmtDn7Od04DQ8) and add some pins in the marlin.
For a sensorless home UART is necessary? If only solder endstop pin and connect to endstop. Is it working? If this is possible how we can add sensitive settings to the GRBL?
[image]

ahmetbozaci, 2022-11-17:
Update

Pascal-m, 2022-11-28:
Great idea

abinoer, 2023-01-07:
Without this function, I/O is insufficient