makerbase-mks/MKS-DLC32 #285: Z-axis problem with MKS-DCL32 V2, possible issue with Z axis pins/circuite on motherboard.

Gerd12, 2022-09-26:
I have tried everything to get the Z-axis to work and home. I have tried the following without any success
I have just added a z-axis for height and focus. For some reason the contrôler/screen (MKS DLC 32 V2 and the touch screen MKS T35-R V2) is showing movement in the negative and positive direction, but the stepper is not moving. I have tried all of the following without success.
1-Tried all move directions and changed speeds, all are fine except Z-axis , no movement of the stepper.
2-verified VREF
3-checked wires and stepper for continuity, everything was OK.
4-changed stepper and driver, did not work. Changed connection to Yt-axis, steppers turned.
5-changed every Z-setting, one at a time in Lightburn, and LASER GRBL with no results.
6-There are no error messages from Lightburn or Laser GRBL.

It seems like there must be an issue with the Z axis pins on motherboard.

If there is someone who can help or make suggestions, this would be really appreciated, and I Thant you all in advance.   G.R.

mrzero8, 2022-12-15:
I also experienced the same thing is there any way to fix this?

Gerd12, 2022-12-15:
I did not find the exact problem, but changed limit switches and wires. Someone suggested a damaged limit switch. I also changed settings in Lightburn to change directions back and forth. Watch your speed and distance settings. I found it was better to change settings on the touch screen for the card. It finally worked. I do not think there was or is a problem with the board.

philvtx, 2023-02-11:
Good morning,
I only have the reference on an X or Y axis, the one who is closest takes control but no Z
I tried several firmware but still the same problem.
anyone have a solution?? thanks in advance
bonjour,
je n'ai la référence que sur un axe X ou Y celui qui est le plus près prend la main mais pas de Z
j'ai essayé plusieurs firmware mais toujours le même problème.
quelqu'un a une solution ?? merci d'avance

Gerd12, 2023-02-11:
Did you update the firmware on your card? The original firmware did not have the z axis control nor the screen for the card, even if it had socket for the wiring. As soon as you install the latest firmware, you will be able to see and control the z axis movement on the touch screen and by the computer. You can get the update on the Makerbase site. Will take some time and passions.

philvtx, 2023-02-11:
hello, thank you for the answer I installed the update but still the same
problem! I tried the 6 versions of the firmware nothing changes on the last
one I no longer have the Z on the screen!
can you tell me which version to put? or the link?
Thank you Best regards

bonjour, merci pour la réponse j'ai bien installé la mise à jour mais
toujours le même problème ! j’ai essayé les 6 versions du firmware rien ne
change sur la dernière je n’ai plus le Z sur l’ecran !
peut-tu me dire quelle version il faut mettre ? ou le lien ?
merci Cordialement

Gerd12, 2023-02-12:
I am running MKS DLC32 CNC v2.0 board with MKS TS35-R V2.0 on V2.1(8M.H35.20220510)The Control screen will show x/y (left, right, top, bottom, with home in the middle) Z up and down with Z home on the side.I do not use SPINDLE and KNIFE selections they are for router and drag Knife machines.Sent from my iPad

philvtx, 2023-02-13:
hello, thank you for the answer I installed the update but still the same
problem! I tried the 6 versions of the firmware nothing changes on the last
one I no longer have the Z on the screen!

can you tell me which version to put? or the link?

Thank you Best regards

 

Bonjour, merci pour la réponse j'ai bien installé la mise à jour mais
toujours le même problème ! jai essayé les 6 versions du firmware rien ne
change sur la dernière je nai plus le Z sur lécran !

Peux-tu me dire quelle version il faut mettre ? ou le lien ?

Merci Cordialement 

 

-- 
Cet e-mail a été vérifié par le logiciel antivirus d'Avast.
www.avast.com

philvtx, 2023-02-14:
Bonjour 
Gerd12 tu parle du paramètre de Lightburn pour changer de direction dans les deux sens. 
Quel est le code stp $? ? Merci 
Je ne comprends pas pourquoi mes références ne se font pas ?

.

Gerd12, 2023-02-14:
Désolé, je ne peux pas vous donner de codes car je ne sais pas comment fonctionne votre machine.  Toute modification peut être effectuée dans les sections de configuration de Lightburn ou Laser Burn.  Les deux fonctionnent avec les fonctions GRBL.  La seule façon d'apprendre est d'essayer.  Bonne chance.Sent from my iPad

philvtx, 2023-02-15:
Merci ça fait 1 mois que j’essaye je n’ai toujours qu’un seul axe qui
prends la référence mais pas de Z !  carte MKS DLC32

Tashadra, 2023-12-29:
@Gerd12 

You could try inverting the enable pins in GRBL with $4=1, that might solve your Z problem but then your X and Y won't work.

Useless, but good as a test.

Same problem I'm having, If only there was an option to chance enable state for each axis.

SHARON2317, 2025-03-05:
I am using this machine and open builds software using but not working when clicking X axis spindle work clicking Y axis spindle stop
can you help me to correct it, if any link for firmware update, please send

Board model MKS DLC 32 V2.1 BOARD

[TWOTREES DIY CNC 3018S Pro Engraver Mini CNC Wood Router GRBL Control Wood router 3 Axis Laser engraving Machine and wood routers] (https://mechblock.in/p/twotrees-ttc3018s-cnc-3018-pro-wood-carving-cnc-router-desktop-woodworking-cnc-router/)