makerbase-mks/MKS-DLC32 #286: MKS-DLC32 v2 motherboard Connection lost.

Gerd12, 2022-09-27:
For some reason the controller shuts down completely after about 1.5 minutes of working. Laser stops, controller stops, and touch screen goes black. After about 2.0 minutes, the motherboard starts again. The touch screen lights up, X and Y axis numbers all returned to 0.0. Is there any way to continue from when / where the CONNECTION was LOST.
Why does the motherboard and screen loose power anyway??  This happens in LIGHTBURN and LASER GRBL.

[image]

Gerd12, 2022-10-07:
Ausgetauscht gegen Neu, Reset und Startknopf.  Im Moment funktioniert alles gut.

danielinkiel, 2022-10-07:
Die Boards sind gut, aber scheinbar sehr empfindlich was Schwankungen angeht (nachvollziehbar). Deshalb gerade bei Bastelei ausreichend dimensionieren.

Viel Erfolg weiterhin.

abinoer, 2023-01-07:
This should be related to the power supply, power cord and other external equipment you use. Please check it yourself

closed 2023-01-07 by abinoer