makerbase-mks/MKS-DLC32 #288: SORT IT OUT

Insigma1, 2022-10-08:
So i have  been waiting 3 months for MKS to sort their dam firmware out, multiple issues reported on the latest version, I am stuck with an older version and frustrated with waiting are you lot going to fix this or what? 

If you're not then can please kindly release the source code so we can this is a joke!

tbsmnea, 2022-10-15:
I see the same way, MKS is getting worse and worse

3Dangel, 2023-01-12:
have you guys tried this?
https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE/tree/main/Firmware/Grbl_Esp32

> So i have been waiting 3 months for MKS to sort their dam firmware out, multiple issues reported on the latest version, I am stuck with an older version and frustrated with waiting are you lot going [...]

closed 2023-03-01 by abinoer