makerbase-mks/MKS-DLC32 #289: how to connect my neje laser head??

djepsylon, 2022-10-08:
hi,
I have a NEJE 30 laser head and the dlc32 card. I searched to find how to connect the head correctly to the motherboard but I found everything and its opposite. Could you once and for all tell me what is the correct way to connect the laser head to the board?
I am attaching an image of the laser in question
thank you.
[image]

djepsylon, 2022-10-08:
i precise that i have a 30A 12V power unit and the stock power unit of the laser. i want to put the laser and the card on the 30A power unit but if it's better or mandatory 2 power supplies I can do it too

abinoer, 2023-03-03:
Sorry, limited energy, not timely reply

Please refer to the instructions or talk to the customer after sales for details

https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf

closed 2023-03-03 by abinoer