makerbase-mks/MKS-DLC32 #292: Reversing X-Y displacements on the TS-35-r display

FMD14, 2022-10-18:
I just installed an MKS DLC32 v 2.1 card with a TS-35-R screen with the latest version CoreXY frimware V2.10_H35_2022_0704_C_ZX_001.bin
On the screen the movements are reversed while on laserGrbl or lightburn and the web application move in the right direction 
How to solve this problem?

manuelchaves, 2022-10-18:
Hello, change the position of the wires in the connectors of the stepper motors

hans-de, 2022-10-18:
Thats a Bug. The Display control xchange the direction. You can live with it, Install an older version with Others Bugs or wait for s fixed version. I decided the 4st Option. Return to sender and by a better board. The Hardware is pretty good, but the Software made me crazy.

FMD14, 2022-10-19:
I just installed version 2.08 the axes work
But I can't configure my limit switches
How to fix this problem
I just requested a refund for the card and screen
Which 32-bit card should I buy for CoreXy
because I feel like MKS does nothing to fix bugs 
They are there to sell and that's all

FMD14, 2022-10-20:
I just asked for the refilling of the card and the screen near the seller is not winning
Which 32-bit card should you turn to have correct results for corexy laser

abinoer, 2023-03-03:
The display moving x backwards has been fixed

closed 2023-03-03 by abinoer

stephendaley, 2025-08-30:
how was it fixed