makerbase-mks/MKS-DLC32 #300: Problem MKS DLC32 with K50 laser co2 no fire

alstare72, 2022-11-19:
Hi there …

 Jai K50 upgraded version of K40
 New power supply and adjustable bed

 I bought mks Dlc32 v2.1 board

 The K50 uses a black box power supply of MYJG60W
 Terminal block different from Ķ40
 FG AC AC ok

 And secondary terminal block H L P G IN 5V

 And I have a 2nd card above the general power supply which feeds the NanoM2 card with Green 24v, Black GND, Red 5V, and Yellow L wires.
 I removed the nano card...
 I took Black and Green to power the MKS ok
 Motor connected ok
 S from MKS to IN
 in lightburn and lasergbl:
 Movement with arrow up down left right ok until ALARM:2

 Lack of co2 laser signal …

 The internet assembly images indicate spindle - to G and Ttl to L

 Ttl on L burns directly at 19ma

 If I add Spindle - to G nothing more
 …

 Laser power problem and lightburn do not want to run my test rectangle ALARM:2
  I tested another assembly that I saw here on ķ40 but still the same the laser burns directly at 20ma continuously as soon as I press the Laser switch
 sorry i use the translator
[image]
[image]
[image]
[image]

abinoer, 2023-01-07:
I don't quite understand what you mean. Is it convenient for you to join QQ group 650031716 for communication

3Dangel, 2023-01-08:
Don't use the TTL to fire the laser. instead use the Spindle's one.
Search for k40 -Mks-DLC32 and you will see.... and only one line. but you would need to power your DLc32 with the same powr supply so the Ground is in common to the laser and to the DLC32. good luck. (also search LB forum using the sear criterial.

> Hi there …
> 
> I have K50 upgraded version of K40 New power supply and adjustable bed
> 
> I bought mks Dlc32 v2.1 board
> 
> The K50 uses a black box power supply of MYJG60W Terminal block different from Ķ40 FG AC AC ok
> 
> And secondary terminal block H L P G IN 5V
> 
> And I have a 2nd card above the general power supply which feeds the NanoM2 card with Green 24v, Black GND, Red 5V, and Yellow L wires. I removed the nano card... I took Black and Green to power the MKS ok Motor connected ok S from MKS to IN in lightburn and lasergbl: Movement with arrow up down left right ok until ALARM:2
> 
> Lack of co2 laser signal …
> 
> The internet assembly images indicate spindle - to G and Ttl to L
> 
> Ttl on L burns directly at 19ma
> 
> If I add Spindle - to G nothing more …
> 
> Laser power problem and lightburn do not want to run my test rectangle ALARM:2 I tested another assembly that I saw here on ķ40 but still the same the laser burns directly at 20ma continuously as soon as I press the Laser switch sorry i use the translator [image] [image] [image] [image]

alstare72, 2023-01-09:
Merci pour votre aide .. 

J'ai changé la carte par avataar120  .. 

Sujet clos.. 
Merci encore

closed 2023-01-09 by alstare72